vshymanskyy/Preferences #4: getType has hardcoded return PT_INVALID

abdasgup, 2022-12-24:
Document (https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/preferences.html) says this method is supposed to check if data exists for the key and return corresponding data type represented by the enum.
But looking into this repo (for ESP8266), it shows the return value is hardcoded to be PT_INVALID. Tested on NodeMCU and it does return so even when I have UChar data saved and can retrieve it using getUChar.

Was the implementation missed while writing it for ESP8266? Can we please have this fixed ASAP?

abdasgup, 2022-12-24:
Looked at the ESP32 implementation (https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/src/Preferences.cpp) and confirmed it's indeed a miss in ESP8266 version.

If original author doesn't get to address it sooner, can someone please create a PR porting the ESP32 implementations equivalent code for this?

vshymanskyy, 2023-03-04:
No, we can't fix this ASAP.

closed 2023-03-04 by vshymanskyy