vshymanskyy/Preferences #6: getString with char * not working.

justbendev, 2023-04-07:
### Hello :wave: 

When porting my current project from ESP32 to ESP8266 all my getString calls didn't work as expected.
Basically when using the char * overload to copy the string into my pointer, 0 bytes are copied to memory.

I will investigate the issue when i have time and send a pull request.

for the time being you can use return value of the String overload and store it that way (Less efficient) if someone is facing a similar issue.

Example code to reproduce the bug
```cpp
#include <Arduino.h>
#include <ArduinoJson.h>
#include <Preferences.h>

Preferences prefs;

char json[] = "{\"p\":\"bTpsL545cVQr7m2AEnQ8GQkrdm1lqH96\"}";

void setup() {
  while(!Serial){
    Serial.begin(115200);
  };

  prefs.begin("example");
  
  if (prefs.getBool("factory",true)){
    Serial.println("--- FACTORY MODE ---");
    StaticJsonDocument <500> doc;
    deserializeJson(doc,json);

    prefs.putString("test","demo");
    prefs.putString("pass",(char *) doc["p"].as<char *>());
    prefs.putBool("factory",false);
    prefs.end();

    ESP.restart();
  }

  Serial.println("--- INITIALIZE ---");
  char PASSWORD[32];

  prefs.getString("pass",PASSWORD,32);
  Serial.println(PASSWORD); 

}

void loop() {
  
}
```

**Have a nice day :wave:**

vshymanskyy, 2024-05-28:
The problem here appears to be the size of the buffer you are reading into.
```cpp
  char PASSWORD[32];
  prefs.getString("pass",PASSWORD,32);
```
The actual data is 32 bytes, BUT the library needs an extra byte for a null terminator.

When enabling the `NVS_LOG`, I actually got a warning about this:
```log
[prefs] WARN: not enough space in buffer: 31 < 32
```

Increasing the buffer helps

vshymanskyy, 2024-05-28:
This was a design decision, `getString` terminates string.
If you don't need string termination, please use `getBytes`.

closed 2024-05-28 by vshymanskyy

vshymanskyy, 2024-05-28:
A small test was added: https://github.com/vshymanskyy/Preferences/commit/cbd5dd581fa7f469140ed246ab7526167cefade4

justbendev, 2024-05-30:
### Hello

Yeah i ended up figuring that out not long after opening the issue 😅 , i just forgot to update it and close it.

Thank you,
and have a great day.

@vshymanskyy