vshymanskyy/Preferences #10: Realtec Ameba BW16 compile error

winterrise, 2023-11-15:
Version: 2.2.1
Date: 2023-08-31T14:35:44.802Z
CLI Version: 0.34.0

Copyright © 2023 Arduino SA

Using Example StartCounter.ino 

WARNING: library Preferences claims to run on all architecture(s) and may be incompatible with your current board which runs on AmebaD architecture(s).
In file included from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\system/component/common/api/platform/platform_stdlib.h:39:0,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\system/component/common/network/ssl/mbedtls-2.4.0/include/mbedtls/config.h:16,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\system/component/soc/realtek/amebad/fwlib/include/rtl8721d_ota.h:176,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\system/component/soc/realtek/amebad/fwlib/include/ameba_soc.h:22,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\system/component/soc/realtek/amebad/cmsis/cmsis.h:35,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\system/component/common/mbed/targets/hal/rtl8721d/PinNames.h:5,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\variants\rtl8720dn_bw16/variant.h:29,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\variants\rtl8720dn_bw16/pins_arduino.h:20,
                 from C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\cores\ambd/Arduino.h:135,
                 from C:\Users\winte\AppData\Local\Temp\arduino\sketches\1B953CE441062D1D68F7CF7E92497711\sketch\StartCounter.ino.cpp:1:
C:\Users\winte\AppData\Local\Temp\.arduinoIDE-unsaved20231015-15716-1ie023u.sw2g\StartCounter\StartCounter.ino: In function 'void setup()':
C:\Users\winte\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.6\system/component/common/api/platform/platform_stdlib_rtl8721d.h:58:22: error: 'class LOGUARTClass' has no member named '_rtl_printf'
  #define printf      _rtl_printf
                      ^
C:\Users\winte\AppData\Local\Temp\.arduinoIDE-unsaved20231015-15716-1ie023u.sw2g\StartCounter\StartCounter.ino:37:10: note: in expansion of macro 'printf'
   Serial.printf("Current counter value: %u\n", counter);
          ^~~~~~

exit status 1

Compilation error: exit status 1

vshymanskyy, 2023-12-06:
Serial.printf is unavailable on Ameba.
Rather easy to fix on the example level. Will do soon.

vshymanskyy, 2024-03-21:
If was fixed ~7 month ago, I forgot to close the ticket

closed 2024-03-21 by vshymanskyy