vshymanskyy/Preferences #15: examples/Prefs2Struct /Prefs2Struct.ino printf

fungreenfox, 2024-05-18:
Esp32 has not "printf"
Please make a version from the Prefs2Struct.ino

vshymanskyy, 2024-05-26:
How comes Esp32 has not "printf".
What do you mean Esp32

fungreenfox, 2024-05-26:
Noticed warnings when trying to use printf.

If you look at all the 3rd party non Arduino.cc products, they all include a printf() method in their Print class.

This allows doing.

Serial.printf("this is an integer counter: %04d", counter);

So if using say ESP32, ESP8266, chipkit, Teensy, Intel, rp2040, STM32,.... etc. platform/cores you have a printf() method.

Arduino.cc absolutely refuses to add a printf() method to the Print class any of their platform/cores.

vshymanskyy, 2024-05-26:
But, how this is esp32 - related? Please clarify your request

fungreenfox, 2024-05-26:
Also https://forum.arduino.cc/t/esp32-where-can-i-find-the-reference-for-serial-printf-the-print-with-the-f-as-a-suffix/1007598

fungreenfox, 2024-05-26:
It appears that printf is not standard on esp32, but might work with adding extra library.

vshymanskyy, 2024-05-26:
This Library is not designed to work on ESP32.
ESP32 does support printf, regardless.

closed 2024-05-28 by vshymanskyy